arduino-libraries/Arduino_MachineControl #73: PMC freeze when using ArduinoBLE with Arduino_MachineControl included.

Bylund, 2021-10-12:
Portenta Machine Control (PMC) crash when the begin function is called for the ArduinoBLE library when Arduino_MachineControl.h is included in the same sketch.

**How to replicate the issue:**
1. Upload the Portenta BLE example without changes (https://docs.arduino.cc/tutorials/portenta-h7/por-ard-ble).
2. Verify functionality. (Optional).
3. Include header file Arduino_MachineControl.h. (#include <Arduino_MachineControl.h>)
4. Upload sketch and note that the Portenta Machine Control will freeze.

**Library versions used:**
- ArduinoBLE v1.2.1.
- Arduino_MachineControl v1.1.0.

gmacario, 2022-07-07:
Hello,

Are there any updates on this subject?

per1234, 2022-11-26:
I'll provide a minimal sketch that will produce the Mbed OS crash reported here:

```cpp
#include <Arduino_MachineControl.h>
#include <ArduinoBLE.h>
void setup() {
  BLE.begin();
}
void loop() {}
```

chminh-ph, 2022-11-26:
Hello,
Remove EncoderClass in Arduino_MachineControl\src\Arduino_MachineControl.h and its object in Arduino_MachineControl\src\Objects.cpp. The issue will be solved. Because of I don't use the Encoder now so It can be a temp solution. I hope it is a key for someone will fix it in ther future.

chminh-ph, 2022-11-26:
Hello,
Remove EncoderClass out of Arduino_MachineControl\src\Arduino_MachineControl.h and its object out of Arduino_MachineControl\src\Objects.cpp. The issue will be solved. Because of I don't use the Encoder now so It can be a temp solution. I hope it is a key for someone will fix it in the future.

rickypid, 2023-10-07:
Hello,
I have the same problem. Are there any updates on this issue?

closed 2023-11-30 by leonardocavagnis