arduino-libraries/Arduino_MachineControl #75: Is there any reason why

mattbue, 2021-11-12:
Hi all,

i'm using the Arduino Machine Control to capture signals from sensors on production machines and send them via USB/serial port to a PC where the data is logged. Usually only the 8 digital inputs are needed.
Since i want to get the signals every millisecond, i realized that reading the signals one by one using
`digital_inputs.read(...)`
takes too much time.
Instead i added methods to the ArduinoIOExpanderClass with which i can use `TCA6424A::readAll(...)` and `TCA6424A::readBank(...)`.

I made some simple measurements by comparing the value comparing from `micros()` before and after running the "read" methods.
I also changed the I2C clock frequency.

`1. Reading channel 0 to 7 one by one @100kHz: 3588 us`
`2. Reading channel 0 to 7 one by one @400kHz: 1337 us`
`3. Reading bank 0 @100kHz: 452 us`
`4. Reading bank 0 @400kHz: 169 us`
`5. Reading bank 0, 1 & 2 one by one @100kHz: 1345 us`
`6. Reading bank 0, 1 & 2 one by one @400kHz: 501 us`
`7. Reading all banks @100kHz: 653 us`
`8. Reading all banks @400kHz: 242 us`

By running I2C at 400kHz and using the `readBank(...)` method I get all 8 bits of bank 0 within ~170 us which is sufficient for me.

Would it be possible to add such methods to ArduinoIOExpanderClass to have them available "out of the box"?

Best regards
matt

pnndra, 2021-11-15:
@mattbue i would suggest you open a PR with your changes.

manchoz, 2022-06-08:
Hi @mattbue, please, check #86

manchoz, 2022-07-26:
Hi @per1234, I think we can close this issue: feature has been implemented and tested and there is no feedback or activity.

per1234, 2022-07-26:
Thanks @manchoz! I'm closing this as resolved by https://github.com/arduino-libraries/Arduino_MachineControl/pull/86

closed 2022-07-26 by per1234