arduino-libraries/Arduino_MachineControl #76: Which is the latest official release of this library?

gmacario, 2021-11-18:
* According to <https://github.com/arduino-libraries/Arduino_MachineControl>, latest release of Portenta Machine Control Library for Arduino is 1.0.2
* According to <https://www.arduino.cc/reference/en/libraries/arduino_machinecontrol/>, latest release of Arduino_MachineControl library is 1.1.0

per1234, 2021-11-18:
Hi @gmacario. Thanks for bringing this to our attention. The latest release is 1.1.0. The reason the repository makes it appear as if the latest release is 1.0.2 is that there are two related things that are treated differently by GitHub:

- [GitHub Releases](https://docs.github.com/en/repositories/releasing-projects-on-github/about-releases)
- [Git tags](https://git-scm.com/book/en/v2/Git-Basics-Tagging)

When a repository has GitHub releases, then the "Releases" page only shows the releases. You must go to the tags page to see the tags.

Since GitHub releases are only a convenience feature, developers sometimes forget to make one after pushing a Git tag. This results in a confusing situation when looking at the repository on GitHub.

I have now created the 1.1.0 release to go with the 1.1.0 tag, so it will show on the Releases page now.

closed 2021-11-18 by per1234

gmacario, 2021-11-18:
Hi @per1234, thanks again for your super-fast reply!