arduino-libraries/Arduino_MachineControl #80: RTD reading too slow

pnndra, 2022-02-07:
the 150 ms delay seems unreasonable. time to select analog muxes shall be much slower

https://github.com/arduino-libraries/Arduino_MachineControl/blob/7697070ed1143df11aac1df8fa7583c1b08cd768/src/Arduino_MachineControl.h#L34

Undermine28, 2022-04-16:
I was just running into this as well. Wondering if there is a way to speed it up?

SJThomas29, 2022-11-24:
I had a go at changing the delay to 100ms and it caused instability in the results, with random spikes to ~500degC. Did also make the function non-blocking similar to the common example 'blink without delay' by having the function return a bool that is true when the change is finished. Can submit a pull request if needed.

closed 2023-09-20 by leonardocavagnis