arduino-libraries/Arduino_MachineControl #85: Dual Core Example/Functionality

will2055, 2022-04-19:
There isn't an example to use the dual core feature of the H7 controller. The current library won't compile for M4 core unless `#ifdef CORE_CM7 ... #endif` are used to contain the RS485 and COM classes.

The Machine Control board is advertised to feature the dual core power of the Portenta H7 controller. However, the current library doesn't allow for that.

elastra21, 2022-05-24:
Does anybody fixed this?, I'm facing the same issue

JustusRijke, 2022-05-27:
same issue here, would be great if @elastra21 fix can be merged to the library

will2055, 2022-05-27:
@elastra21 Could you provide a brief summary of what changes you made for this fix?

elastra21, 2022-05-27:
Yeah as you commented at the start, I just change this ['#ifdef CORE_CM7](https://github.com/elastra21/Arduino_MachineControl/blob/cbbae20bdf1420e43912ce167760b7a49ad166cd/src/Arduino_MachineControl.h#L87-L157) in Arduino_MachineControl.h

manchoz, 2022-07-26:
Hi @elastra21, would you mind creating a PR from your proposed changes? This will help discussing your proposed changes and eventually merging them.

elastra21, 2022-10-26:
> Hi @elastra21, would you mind creating a PR from your proposed changes? This will help discussing your proposed changes and eventually merging them.

PR submitted.