arduino-libraries/Arduino_MachineControl #95: Analog output malfunction on AO2

i-herrera, 2022-08-03:
It seems like AO2 pin is not correctly configured for PWM. There must be an issue in the AnalogOutClass section of the Arduino_MachineControl lib:
https://github.com/arduino-libraries/Arduino_MachineControl/blob/master/src/Arduino_MachineControl.h#:~:text=class%20AnalogOutClass%20%7B

The issue was discovered as follows: Even when AO2 is set to output 0V it always outputs 10V. I managed to replicate this result using the 'Analog_Out' sketch from Arduino_MachineControl library and configuring the Ao" as follows:

inside setup(): analog_out.period_ms(2, 4);
in loop(): analog_out.write(2, 0);

bradmartin333, 2023-02-08:
I was able to set AO2 to different values, but the initial value for this IO is 10V
[Here is a video I made demonstrating this](https://youtu.be/RRycd1lvnMI)

leonardocavagnis, 2023-09-27:
Hi @i-herrera, @bradmartin333,
This issue is caused by an erroneous management of the mbed-os core for the High-Resolution timer connected to AO2 pin. 
Refer to the linked PR.

closed 2023-09-29 by facchinm