arduino-libraries/Arduino_MachineControl #104: PT1000 isn´t compatible

AceTecNic, 2023-02-09:
in Temp_probes_RTD.ino you describe that a PT1000 is also possible. Now your support writes me that this is not possible. Please take that out of the comment in the code, just causes confusion. Thanks :)

MDBInd, 2023-03-23:
Like you, I was advised on 08/02/23 that "Unfortunately at the moment PT1000 are not supported" by Arduino support. It is quite confusing as PT1000 is printed on the board. Hopefully it will be supported in the future.

MDBInd, 2023-05-10:
Quick update, in the latest version of the [documentation](https://docs.arduino.cc/resources/datasheets/AKX00032-datasheet.pdf) (Updated 08/05//2023) all references of PT1000 have been removed.

leonardocavagnis, 2023-09-20:
Hi @AceTecNic, @MDBInd 
I confirm that it is not compatible. All references to pt1000 have been removed.

closed 2023-09-20 by leonardocavagnis