arduino-libraries/Arduino_MachineControl #105: digital_programmables.init()

Sarah26-10, 2023-02-17:
I have a huge problem with the digital_programmable pins. When i have 24 V connected the board crashed when running the .init() function. But when there is no 24V the function runs without problem. 
I tried many things, like another sequence of functions in the setup, but i doesn`t work. 
Does anyone has the same problems or an idea?
A code example does not help here, because i can move the digital_programmables.init() at any position in the setup().

Please,help :)

bradmartin333, 2023-03-14:
Hmm haven't run into that. Here is my setup() - maybe you're missing something in sequence?

``` cpp
void setup() {
  Serial.begin(9600);
  // Start UDP
  Ethernet.begin(ip);
  Udp.begin(localPort);
  IPAddress ip = Ethernet.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);
  // Setup APMC
  Wire.begin();
  // J3: Digital Inputs
  digital_inputs.init();
  // J6: Digital Outputs
  digital_outputs.setLatch();
  digital_outputs.setAll(0);
  // J8: Programmable Digital I/O
  digital_programmables.init();
  digital_programmables.setLatch();
  // J11: Analog Out
  //For reference: analog_out.period_ms(CHANNEL, PERIOD_MILLISECONDS);
  analog_out.period_ms(0, 4);
  analog_out.period_ms(1, 4);
  analog_out.period_ms(2, 4);
  analog_out.period_ms(3, 4);
  // J9: Analog In
  analogReadResolution(16);
  analog_in.set0_10V();
}
```

leonardocavagnis, 2023-09-20:
Hi @Sarah26-10, 
I'm unable to reproduce the problem you've described. It's possible that the issue may have been specific to your environment or a temporary hardware problem.

closed 2023-09-20 by leonardocavagnis

Sarah26-10, 2023-09-20:
Yes as an explanation for my issue: It was my fault. There was a problem in combination of the digital_programmables.init() with I2C initialisation and sending of commands and the series in which this happens. So in case someone has the same problem check your routine and sequence of execution.

Thanks :)