arduino-libraries/Arduino_MachineControl #116: "Device attach failed" USB Host in the portenta machine control

Vasanth28897, 2023-12-20:
I am connected a USB-UART connector from the bms to the USB-A port in the PMC, and upload this `USB_Host.ino` example code. The serial monitor output shows like this
```
RHUB FS:0      Enable
DEV  FS.00     Allocate pipe, in = 0, out = 1
DEV  FS.01     Re-allocate pipe, in = 0, out = 1
DEV  FS.01     FS device, VID:1a86 PID:7523
ITF  FS.01.0-0 Vendor interface init
ITF  FS.01.0-0 Ep Info: 3 eps, in_ep:[82,2,32], out_ep:[02,2,32],
DEV  FS.01     Device has 1 interfaces
RHUB FS:0      Device attach failed
```
whenever i unplug the connector from the USB-A port, the output in the serial monitor changes like this...
```
RHUB FS:0      Disable 4
ITF  FS.01.0-0 Vendor interface deinit
RHUB FS:0      Disconnect
RHUB FS:0      Connect
RHUB FS:0      Already enabled
RHUB FS:0      Enable
DEV  FS.00     Allocate pipe, in = 0, out = 1
DEV  FS.01     Re-allocate pipe, in = 0, out = 1
DEV  FS.01     FS device, VID:1a86 PID:7523
ITF  FS.01.0-0 Vendor interface init
ITF  FS.01.0-0 Ep Info: 3 eps, in_ep:[82,2,32], out_ep:[02,2,32],
DEV  FS.01     Device has 1 interfaces
RHUB FS:0      Device attach failed
```
what do i have to do to make a successful connection?

leonardocavagnis, 2023-12-22:
Hi @Vasanth28897 
What type of USB device are you connecting? what is BMS?

Vasanth28897, 2023-12-22:
I am trying to establish a serial over USB communication between Arduino PMC and BMS(Battery Management System) to get the battery data. So i am connecting the USB-UART connector from the BMS in the USNB-A port in the PMC. 
How Can I do that by USB-A port in the PMC act as a USBhost?

leonardocavagnis, 2023-12-22:
Ok, try to use [USBHostSerial](https://github.com/arduino-libraries/Arduino_USBHostMbed5/tree/main/src/USBHostSerial) class of the [Arduino_USBHostMbed5](https://github.com/arduino-libraries/Arduino_USBHostMbed5) library.

Vasanth28897, 2023-12-23:
thanks, i will update here.

Vasanth28897, 2023-12-23:
tried that one, didn't work out. Also using that class included in the program, the board went to bootloader mode everytime upload.

Vasanth28897, 2024-01-03:
The USB-UART converter module does not compatable with Arduino Portenta H7 board - from Arduino Support. So, i am closing this issue as not planned.

closed 2024-01-03 by Vasanth28897